InfernoEmbedded/PowerScraper #1: how to use

sommo, 2018-07-25:
Hi!
i'm back :P i see you update the project,
how could i use it?
after edit the config.tml (but what i have to edit?) i was thinking that i need to edit inverter's ip address but can't find it...
could you help me?
Thank you!

deece, 2018-07-25:
If you want to use the Wifi interface, add a list of invert IPs or hostnames under 'inverters', in the sample, there are 3 hostnames there.

I've added support for the ethernet interface too, to user that, set the inverters list for the Wifi section to the empty list '[]', and add the IP address or hostname of the ethernet interface under 'inverters' for the Solax-Modbus section.

If you don't have an SDM630 Power meter, you should leave the 'ports' line of that section as the empty list too.

deece, 2018-07-25:
I spoke to the Solax engineers, there is no way to disable the Wifi interface. You can significantly reduce the signal though by removing the top and front cover, then disconnecting the antenna cable from the motherboard. This will probably void your warranty though, so do this at your own risk.

sommo, 2018-07-25:
All clear, I think tomorrow I will try!
My inverter is connect via wifi to my my network, so i will put the inverter ip where u said, or by wifi u meaning when connecting to inverter ap wifi?
Any way for the enomcms side, I need to configure sensor right?
Like in other post says? Or needs a custom configuration for them to work with your script?
It could be awesome also to trigger the collected data not through enomcms or any other external service/cloud but directly to a webserver in lan (so i can get graphs also if there is no internet connection or cloud maintenance) and most important no data goes out from our home, at least we could publish it directly from our public ip (on specific port) to reach it when we are away..
What you think?

deece, 2018-07-25:
1. It needs your inverter's IP on your Wifi interface.
2. Once working, you should see inputs show up in EmonCMS from your inverter, you can then create feeds based on this.
3. The software is modular, so if you want to stash the data elsewhere, you can write an output module to do that. I would just run EmonCMS locally though, it's not hard to set up.

sommo, 2018-07-26:
Amazing!
I’m looking to install emoncms on my pi, didnt thought it could be possible.
Thank you!
I will let you know!

Inviato da iPhone

sommo, 2018-07-30:
Hi!
i'm having issue when trying to launch the script:

=> python power_scraper.py
Traceback (most recent call last):
  File "power_scraper.py", line 37, in <module>
    from Inputs.SolaxWifi import SolaxWifi
ImportError: No module named Inputs.SolaxWifi

my config.toml:
poll_period = 10 # seconds

[Solax-Wifi]
timeout = 5 # seconds
inverters = ['192.168.0.106']

[Solax-Modbus]
timeout = 5 # seconds
inverters = []

[SDM630Modbusv2]
timeout = 5 # seconds
baud = 38400
parity = 'E'
stopbits = 1
ports = []

[emoncms]
timeout = 5 # seconds
server = "http://xxxxxxxxx"
api_key = "xxxxxxxxxxxxxxxx"

all dependecies seems ok:
sudo pip install toml twisted pymodbus
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: toml in /usr/local/lib/python2.7/dist-packages (0.9.4)
Requirement already satisfied: twisted in /usr/lib/python2.7/dist-packages (16.6.0)
Requirement already satisfied: pymodbus in /usr/local/lib/python2.7/dist-packages (1.5.2)
Requirement already satisfied: pyserial>=2.6 in /usr/local/lib/python2.7/dist-packages (from pymodbus) (3.2.1)

deece, 2018-07-30:
How did you download the program? If you just downloaded the script & config file, you'll need to clone the full repo (or download & extract the archive).

deece, 2018-07-30:
Hmm, and are you running it from the directory you cloned it into?

sommo, 2018-07-30:
yes of course, otehrwise i could not execute it:

pi@emonpi(rw):PowerScraper$ pwd
/home/pi/PowerScraper

pi@emonpi(rw):PowerScraper$ python power_scraper.py
Traceback (most recent call last):
  File "power_scraper.py", line 37, in <module>
    from Inputs.SolaxWifi import SolaxWifi
ImportError: No module named Inputs.SolaxWifi

don't know if emonpi distro, has some issue...
i had error by installing via pip some dependaince, but did

in fact, i can't execute it from the wrong folder:
pi@emonpi(rw):~$ python power_scraper.py
python: can't open file 'power_scraper.py': [Errno 2] No such file or directory

the error is:

ImportError: No module named Inputs.SolaxWifi

so it seems a script problem.. don't know to trust the emonpi distro but i think is usefull to run your script and the emoncms server on the same machine.
i could try to install the script on a different rpi if you don't know how to resolve or point me to that error.

deece, 2018-07-30:
Could you please paste the output of `find .` from the directory with the script? We should see 'Inputs/SolaxWifi.py' in there.

deece, 2018-07-30:
One other thing to try, I've only tested with Python 3.6, not 2.7. I'm still fairly new to Python, but there may be some differences there.

sommo, 2018-07-30:
i'm struggling with installing python3 and it's dependancy, no way still haveing problem installing pymodbus.... i can't figured out.. so can't test with python 3.x

sommo, 2018-07-30:
maybe is the python version but don't know

sommo, 2018-07-31:
installed python3 on a mac, i recieve error anyway:
python3 power_scraper.py 
Traceback (most recent call last):
  File "power_scraper.py", line 38, in <module>
    from Inputs.SolaxModbus import SolaxModbus
  File "/Users/sommo/Desktop/Solax/PowerScraper/Inputs/SolaxModbus.py", line 1
    from pymodbus.client.async import ModbusClientProtocol, ModbusClientFactory

there's something else..

sommo, 2018-07-31:
> Do you have PYTHONPATH set in the environment? If so, you may need to add '.' or the path to the powerscraper directory to it.

What’s the command to do so?

Inviato da iPhone

deece, 2018-07-31:
`echo $PYTHONPATH` will tell you if it's set and what the value is
`export PYTHONPATH="$PYTHONPATH:/path/to/powerscraper"` will add the powerscraper path to it

sommo, 2018-08-01:
it doesn't help, thank you anyway

deece, 2018-08-13:
Have you had any luck? If not, try changing "from Inputs" to "from .Inputs" (& the same for outputs) as described here, and let me know how you go: https://stackoverflow.com/questions/24722212/python-cant-find-module-in-the-same-folder

averter, 2018-10-09:
@sommo I am trying to have the same setup as you. However I am still stuck trying to access the data. Did you configured anything special on your LAN to access the inverter? When I try to access http://ipinverter/api/realTimeData.htm or http://ipinverter/ I am met with a login webpage from Solaxpower. I cannot logon to it with the default admin/admin. Thanks for any information

deece, 2018-10-09:
@averter i wouldn't bother with the wifi interface. Modbus on the wired Ethernet interface is a lot richer, and allows you the option of managing the battery by software.

closed 2018-11-15 by deece