InfernoEmbedded/PowerScraper #5: PowerScraper on Solax X1

PAk-CatchFire, 2019-06-19:
Hello. I would like to know if this would work on the 485 modbus port othe Solax X1 inverter.

Thank you and great work!!!

deece, 2019-06-19:
Probably, but I haven't tested it, so I can't say 100% for sure.

It shouldn't hurt to try though (First just try monitoring rather than battery control).

PAk-CatchFire, 2019-06-19:
Thank you!!!

But, do I need anything else on the 485 bus? I was hoping to port your code to a raspberry (or ESP32 in C) and communicate and give only read parametres.

Regards

deece, 2019-06-19:
Nope, it should run as-is on a raspberry pi, that's how I run mine (on the same machine as EmonCMS). I just have a USB->RS485 widget to interface the pi to the inverter.

deece, 2019-06-19:
If you don't configure Solax-batteryControl in the config, the script won't write to any registers.

PAk-CatchFire, 2019-06-19:
Awesome!!!
Is it possible to ask the inverter one parameter by one? Do you have an example of the RS485 serial packet? 

On the other hand, do you have a register map of the solax inverter? Where did you get all that info?
I was searching for it but couldn't find anything.

I read somewhere that this inverter only sends data on demand.

Sorry for all the questions.

deece, 2019-06-20:
Yes, have a look at Inputs/SolaxModbus.py. I use a third party Modbus library, but the Modbus protocol is well documented. The call to read_input_registers is (start address, number of sequential registers to read). The offsets in solaxRegisterCallback are relative to the start address.

You can bug a Solax engineer for the read-only register map - I can't share the documentation I have unfortunately.

PAk-CatchFire, 2019-06-20:
Thank you @deece 
I hecked SolaxModbus.py, but in that file you use: `reactor.connectTCP` and I thought it was a simple RS485 connection, not a TCP socket.

I thought I had to use the Modbus/RTU (RS485) protocol from the file SDM630Modbusv2.

Am I wrong?

deece, 2019-06-20:
Oh, whoops, I got confused. I use Modbus/RTU to speak to my power meter. The communication with my inverters is Modbus/TCP.

I expect the registers will be the same regardless of the transport.

deece, 2019-06-20:
the SDM630ModbusV2 is an external power meter, it's not related to the Solax inverter communications.

PAk-CatchFire, 2019-06-20:
Ok, I see.
Would you mind to make a little test on the 485 bus doing the same you do on the TCP port?

Regards

deece, 2019-06-20:
My machines don't have RS485 ports, sorry

closed 2019-10-02 by deece