InfernoEmbedded/PowerScraper #6: Mapping Register Values

wills106, 2019-10-01:
I have managed to take your register values from Inputs/SolaxModbus.py and convert them into decimal values to use in Home Assistant as package configuration.

https://github.com/wills106/homeassistant-config/blob/master/packages/solax.yaml

I have worked out some of the values for Run Mode (0x09)
The following are decimal results I have worked out.

0 - Waiting
1 - Checking
2 - Normal Mode
9 - Idle Mode

Have you managed to figure out any values for the following?
Battery Energy Charged (0x1D)
Battery Energy Discharged (0x20)
Battery State of Health (0x23)
Solar Energy Total (0x70)

Solar Energy Total in particular has stayed at 0 for the whole time.

Thanks

deece, 2019-10-02:
Battery Energy charged/discharged is the amount of energy to/from the battery in kWh.

State of Health is the health of the battery (reported as a percentage)

Solar Energy Total is supposed to be the total energy from the panels in kWh, but my graphs don't line up with that - it is non-zero though, but has been constant for the past few months.

closed 2019-10-03 by deece

wills106, 2019-11-11:
Incase it helps anyone I have figured out another two run modes:

3 - Off Mode
7 - EPS Mode