InfernoEmbedded/PowerScraper #8: TL5000C?

glynhudson, 2020-02-26:
As anyone tested this to read data from a TL5000C inverter? And a Triple-power HV10045 battery?

deece, 2020-03-01:
I think you're the first, it would be nice to know if the TL series work. I'll leave this open so you can report your findings.

wills106, 2020-05-22:
@glynhudson 
The newer Hybrid inverters don't use remote mode.
I have got them working with Home Assistant based on the info I learnt from PowerScraper.

I would start of with solax_x1_hybrid_g3_triplepower.yaml as the pylontech version uses alot higher charge amp's compared to the Triple Power due to being lower voltage.
The registers are converted into decimal in my example.

https://github.com/wills106/homeassistant-config/tree/master/packages

https://community.home-assistant.io/t/solax-inverter-by-modbus-no-pocket-wifi/140143/45

closed 2021-08-05 by deece