InfernoEmbedded/PowerScraper #9: ModuleNotFoundError: No module named 'toml'

glynhudson, 2020-02-29:
I'm having trouble running the script, I'm getting an error loading the `toml` module even though python pip is showing that it's correctly installed. Any ideas what I'm missing? Thanks 

```
./power_scraper.py 
Traceback (most recent call last):
  File "./power_scraper.py", line 25, in <module>
    import toml
ModuleNotFoundError: No module named 'toml'
```

```
pip freeze
asn1crypto==0.24.0
attrs==19.3.0
Automat==20.2.0
certifi==2019.9.11
chardet==3.0.4
colorzero==1.1
configobj==5.0.6
configparser==3.5.0b2
constantly==15.1.0
cryptography==2.6.1
entrypoints==0.3
enum34==1.1.6
gpiozero==1.5.1
hyperlink==19.0.0
idna==2.8
incremental==17.5.0
ipaddress==1.0.17
keyring==17.1.1
keyrings.alt==3.1.1
meld3==1.0.2
paho-mqtt==1.4.0
pycrypto==2.6.1
PyGObject==3.30.4
PyHamcrest==1.10.1
pymodbus==2.3.0
pyserial==3.4
pyxdg==0.25
redis==3.3.11
requests==2.22.0
RPi.GPIO==0.7.0
SecretStorage==2.3.1
six==1.12.0
spidev==3.3
supervisor==3.3.5
toml==0.10.0
Twisted==19.10.0
urllib3==1.25.6
xmltodict==0.12.0
zope.interface==4.7.1
```

glynhudson, 2020-03-01:
Solved: the issue was due to python 2 being installed as standard and PowerScraper requires python3. Installing pip3 and installing the modules fixed the issue:

```
apt-get install python3-pip `
pip3 install toml twisted pymodbus
```

closed 2020-03-01 by glynhudson

Kelevraquackenstein, 2024-08-25:
@glynhudson 

Thank you for taking the time to share this. 

This solved my problem.