InfernoEmbedded/PowerScraper #10: SyntaxError: invalid syntax?

glynhudson, 2020-03-01:
I've now up and running with python 3.7.3 and all the modules installed, however running under python3 I get the following error. Any idea what I'm missing?

```
python3 power_scraper.py 
Traceback (most recent call last):
  File "power_scraper.py", line 38, in <module>
    from Inputs.SolaxModbus import SolaxModbus
  File "/opt/PowerScraper/Inputs/SolaxModbus.py", line 1
    from pymodbus.client.async import ModbusClientProtocol, ModbusClientFactory
                             ^
SyntaxError: invalid syntax
```

```
pip3 freeze
asn1crypto==0.24.0
async==0.6.2
attrs==19.3.0
Automat==20.2.0
certifi==2018.8.24
chardet==3.0.4
constantly==15.1.0
cryptography==2.6.1
entrypoints==0.3
hyperlink==19.0.0
idna==2.6
incremental==17.5.0
keyring==17.1.1
keyrings.alt==3.1.1
pycrypto==2.6.1
PyGObject==3.30.4
PyHamcrest==2.0.1
pymodbus==2.3.0
pyserial==3.4
python-apt==1.8.4
pyxdg==0.25
requests==2.21.0
SecretStorage==2.3.1
six==1.12.0
ssh-import-id==5.7
toml==0.10.0
Twisted==19.10.0
ufw==0.36
urllib3==1.24.1
zope.interface==4.7.1
```

deece, 2020-03-01:
I think I wrote this with an older version of pymodbus. Try version 1.5.1

glynhudson, 2020-03-03:
Thanks for reply, I downgraded to 

`pip3 install pymodbus==1.5.1`

```
pi@emonpi:/opt/PowerScraper $ pip3 freeze
asn1crypto==0.24.0
async==0.6.2
attrs==19.3.0
Automat==20.2.0
certifi==2018.8.24
chardet==3.0.4
constantly==15.1.0
cryptography==2.6.1
entrypoints==0.3
hyperlink==19.0.0
idna==2.6
incremental==17.5.0
keyring==17.1.1
keyrings.alt==3.1.1
pycrypto==2.6.1
PyGObject==3.30.4
PyHamcrest==2.0.1
pymodbus==1.5.1
pyserial==3.4
python-apt==1.8.4
pyxdg==0.25
requests==2.21.0
SecretStorage==2.3.1
six==1.12.0
ssh-import-id==5.7
toml==0.10.0
Twisted==19.10.0
ufw==0.36
urllib3==1.24.1
zope.interface==4.7.1
```

However, I get the same error, have you got any other ideas?:

```
pi@emonpi:/opt/PowerScraper $ ./power_scraper.py 
Traceback (most recent call last):
  File "./power_scraper.py", line 38, in <module>
    from Inputs.SolaxModbus import SolaxModbus
  File "/opt/PowerScraper/Inputs/SolaxModbus.py", line 1
    from pymodbus.client.async import ModbusClientProtocol, ModbusClientFactory
```

commit a5e886c6 (2020-03-20): update to work with PyModbus 2.3.0 fix #10

https://community.openenergymonitor.org/t/solution-reading-solax-sk-su5000-e-inverters-sdm630v2-modbus-power-meters/8214/80?u=glyn.hudson
closed 2020-03-20 by deece