InfernoEmbedded/PowerScraper #17: ModuleNotFoundError: No module named 'pymodbus.client.asynchronous'

MrHappy, 2021-08-23:
I ran a few updates today and ran into an error message.
This error appeared for SolaxModbus as well as SolaxXHybridModbus.
I use the Solax-Wifi myself and this one does not give me any data anymore.
(I haven't monitored my system in detail lately, so I'm not sure whether the lack of data is a result of the solarpanels not producing any energy or a result of the error below)
```
/usr/bin/python3 /home/emoncms/PowerScraper/power_scraper.py
Traceback (most recent call last):
  File "/home/emoncms/PowerScraper/power_scraper.py", line 38, in <module>
    from Inputs.SolaxModbus import SolaxModbus
  File "/home/emoncms/PowerScraper/Inputs/SolaxModbus.py", line 2, in <module>
    from pymodbus.client.asynchronous.twisted import ModbusClientProtocol
ModuleNotFoundError: No module named 'pymodbus.client.asynchronous'
```

deece, 2021-08-23:
You'll need to ensure all the prerequisites are installed. I've started compiling a list of them in Readme.md

MrHappy, 2021-08-24:
> You'll need to ensure all the prerequisites are installed. I've started compiling a list of them in Readme.md

I have those installed. Only thing related  I could find on google was that the module is deprecated.
Disabling both inputs in power_scraper.py lets it start, but does this also let the wifi part work?

deece, 2021-08-24:
The new input in EmonCMS is unrelated - I've seen similar issues in the past, which were resolved by removing the duplicated inputs, then flushing the Redis cache.

deece, 2021-08-24:
FYI I'm using pymodbus 2.1.0. You can try pip3 install pymodbus==2.1.0 to force an older version.

MrHappy, 2021-08-24:
I remarked both inputs, changed the feeds to use the new node and everything seems to be up and running again,

MrHappy, 2021-08-24:
Btw. Do you know what the different status-codes stand for?

deece, 2021-08-24:
As reported through the Wifi interface? No, I never received any documentation on those and gave it up when I discovered the Ethernet interface speaks Modbus & provides much richer data.

I wouldn't be surprised if it's the same as the inverter fault code register.

[image]

MrHappy, 2021-08-25:
This is the line I get on the api/realtimedata.htm
`{"method":"uploadsn","version":"Solax_SI_CH_2nd_20160912_DE02","type":"AL_SI","SN":"9999999","Data":[3.2,3.3,369.6,321.9,9.4,236.1,2176,38,23.5,10423.2,0,1205,1077,,,,,,,,,,,,,,,,,,,,,,,,,,,,,0.00,0.00,,,,,,,,49.97,,,0.0,0.0,0,0.00,0,0,0,0.00,0,8,0,0,0.00,0,8],"Status":"2"}`
It looks like it always returns Status:2. I suppose this can change, but what does it mean? (And what alternative values are there?)

closed 2025-04-12 by deece